lancaster-university/codal-nrf52 #6: Linking error : multiple definition

OpusK, 2020-02-04:
Hi, 

When building the sync_serial branch, I got the following error:
This problem is due to the definition of IRQHanlder in nrfx_prs.c by setting template/nrfx_config.h and the same IRQHanlder in your sync_serial.cpp.

Modifying nrfx_config.h simply solves the problem. However, since it is located in codal-nrf52, it cannot be managed by nrf52840-dk.

In the CODAL structure, core/nrf52/nrf52840-dk are compiled separately and built by linking each other. After all, when compiling nrf52, nrfx_config.h is included, so the file must be placed in codal-nrf52.

Is there a reason for this design? Or can you please let me know if there is a good solution?

```bash
[ 72%] Built target codal-nrf52840-dk
[ 72%] Linking CXX executable NRF52840_DK
libcodal-nrf52.a(nrfx_prs.c.o): In function `SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQHandler':
/src/libraries/codal-nrf52/nrfx/drivers/src/prs/nrfx_prs.c:60: multiple definition of `SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQHandler'
libcodal-nrf52.a(sync_serial.cpp.o):/src/libraries/codal-nrf52/source/sync_serial.cpp:73: first defined here
libcodal-nrf52.a(nrfx_prs.c.o): In function `SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQHandler':
/src/libraries/codal-nrf52/nrfx/drivers/src/prs/nrfx_prs.c:60: multiple definition of `SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQHandler'
libcodal-nrf52.a(sync_serial.cpp.o):/src/libraries/codal-nrf52/source/sync_serial.cpp:35: first defined here
libcodal-nrf52.a(nrfx_prs.c.o): In function `SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0_IRQHandler':
/src/libraries/codal-nrf52/nrfx/drivers/src/prs/nrfx_prs.c:60: multiple definition of `SPIM2_SPIS2_SPI2_IRQHandler'
libcodal-nrf52.a(sync_serial.cpp.o):/src/libraries/codal-nrf52/source/sync_serial.cpp:35: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [NRF52840_DK] Error 1
make[1]: *** [CMakeFiles/NRF52840_DK.dir/all] Error 2
make: *** [all] Error 2
```

OpusK, 2020-02-04:
Oh, my question was foolish.

I can do this with the processor's preprocessing.

closed 2020-02-04 by OpusK

finneyj, 2020-02-04:
ah, cool. Thanks for reporting back @OpusK!

There's no such things as a foolish question... it's all just part of the journey. ;)